supermerill/SuperSlicer #3575: Dual Filament Issues

utzy79, 2023-01-15:
### Version 2.5.59
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version Windows 11
_What OS are you using, and state any version #s_

### Behavior
Multi Material Prints not working correctly

When slicing a multi material print that has been setup from the Multimaterial printing option, 
It still leaves parts of the print using the wrong tool

Can confirm that the same file works in Prusa Slicer

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

[image]
[image]

mjfsch, 2023-01-15:
under print settings, slicing, change the resolution to 0, this fixed it for me

[image]

utzy79, 2023-01-16:
Thank you, this got rid of most of it but left it in a couple of spots, then did a simplify model to cause a remesh and it worked.

closed 2023-01-16 by utzy79