supermerill/SuperSlicer #3583: Support Option: Join Distance like in Cura

andilee7, 2023-01-19:
**Is your feature request related to a problem? Please describe.**
When printing many small supports they get fragile at a specific height and one support tower can fall over.

**Describe the solution you'd like**
A option like in Cura to specify a join distance. 

**Describe how it would work**
When supports which are closer together than that specified distance, the supports gets joined into one big support element.

**Describe alternatives you've considered**
Tried to paint supports on the areas which doesn't require support to force join them. But this also didn't created support probably due to the overhang threshold

**Additional context**
[image]
(Image from here: https://all3dp.com/2/cura-support-settings-optimize-your-supports/ )

In a real life sample, two of those towers fell over after 10 hours
[image]

neophyl, 2023-01-19:
What version are you using ?  Where is your attached project file with an example ? (File>Save Project As).

All painting or a modifier do is mark surfaces on the MODEL that you think need support (or don't with the blocker).  The support generation algorithm then works with what you have defined, with the settings it has been configured with.   They have never been 'put support here' or 'don't allow support to pass though this area' tools.

If using 2.5.x based on PS 2.5 then supports now have 2 options, the old grid supports and the newer Snug option.  If using Snug then there is a settings called Closing Radius which does exactly that.  By default it is set to 2mm.  You might want to experiment with that setting.  It might still not do what you want (couldn't check without your project though).

andilee7, 2023-01-20:
Thanks for your fast reply! I'm using 2.5.59 and just tried your suggestion with snug support and the closing radius in combination with paint on support and I it looks pretty promising in the preview. Maybe a bit too much support material but better than the small towers. Thank you very much for your help!

**Update before sending the comment**: I just tried exporting you the project in case you want to look at it and now I was able to achieve the same thing with grid support and paint-on supports. Maybe I've not painted good enough the last few times I've tried it. I don't know. I think we can close this feature request as I'm now able to create it with snug and grid support in combination with a good paint job 😄 

Attached are 3 projects for your reference if you still want to have a look. 
- Snug support + higher value of closing radius + paint-on supports in the critical areas -> lots of support but should do the job
- Grid support + paint-on supports in the critical areas -> Represents the ideal solution which I wasn't able to create before posting this feature request
- Grid support without additional paint-on -> The small support towers I wanted to avoid

[projects.zip](https://github.com/supermerill/SuperSlicer/files/10463414/projects.zip)