supermerill/SuperSlicer #3588: Continuous Perimeter Required

swpaxton66, 2023-01-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Perimeters are split for overlapping models.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Would like to have 1 continuous perimeter for all imported models so the outline is smooth on the first layer (and other layers if required) (This is how it works in Simplify3D)

**Describe how it would work**
Be as descriptive as your skill level allows you.
Import multiple models. Overlap at least 2 models.  The perimeter should flow around both objects and not completed on each object separately.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I've done some SuperSlicer internet research and tried changing perimeter settings.  No luck.

**Additional context**
Add any other context or paint drawing about the feature request here.
[image]

[image]

swpaxton66, 2023-01-21:
Found it as merge but the perimeters are a bit weird 
[image]

swpaxton66, 2023-01-21:
[image]
[image]

neophyl, 2023-01-21:
Without your project file its going to be impossible to know.  Please attach a zipped up project file  (File>Save Project As).  

Also would be helpful if you listed which version you were using.  You know, all the basic info that's asked for when opening a new issue.

swpaxton66, 2023-01-21:
Thank you for the prompt response.  Yeah makes perfect sense.  I was blinded by a million things on the go, blindly following along the questions from the boiler plate issues list.  Here is the project file.  Would it be possible to add all of said requirements to the list of questions?  (Might be good to have it right in front of the user's face).

Windows 10
[Merged.zip](https://github.com/supermerill/SuperSlicer/files/10473278/Merged.zip)

SuperSlicer 2.4.58.5

neophyl, 2023-01-21:
You have Perimeters & Shell>Quality>Only one Perimeter ticked for On top surfaces.  Those helper disks are one layer thick and are top surfaces.  As such its applying that to the entire shape which is what is pushing the inner perimeters around it.

swpaxton66, 2023-01-21:
Got it.  Thanks for investigating.

closed 2023-01-21 by swpaxton66