supermerill/SuperSlicer #3591: Add some kind of way to run/make plugins or scripts (like cura)

GuySandler, 2023-01-22:
**Is your feature request related to a problem? Please describe.**
I want to be able to add features like tree supports or just custom things in general, but prusaslicer isn't getting it anytime soon

**Describe the solution you'd like**
Add a plugin system like cura or at least a way to run scripts

**Describe how it would work**
like cura, you make a plugin, but I guess in C++ here or you can run scripts in C++ to do stuff

**Describe alternatives you've considered**
I have considered switching entirely to cura but I still want some of the good features in prusaslicer like the text they are about to add

**Additional context**
non

mjfsch, 2023-01-23:
Not sure if its exactly what your after, but superslicer does have the ability to run scripts, see the Generate menu for running FreeCad Python scripting.

Also check out the output options section of the Print Settings menu, there is the ability to run post processing scripts

inspectionsbybob, 2023-01-26:
There is also the "Post Processing" section that I use for running "ArcWelder" which works very well. 
(once you figure out the full path to your script)