supermerill/SuperSlicer #3592: Infill not allways connected to infill

itchy2, 2023-01-22:
### Version
2.4.58.5

### Operating system type + version
Windows 10 21H2

### Behavior
When printing infill, there are gaps / holes in it. It looks like by crossing the infill lines it stopes extruding a little bit to early / starts to late. Tried different print settings / profiles / filaments to understand it but there is always the same issue. It happens in different ways on all type of infills. EM is calibrated with Superslicer style and validated by using Voron Pads and all thingness measurement. Overall printing results look fine and without issues, just the infill is not working as expected. I am pretty sure that it has to do something with my settings in Superslicer but I have no clue which settings. When printing with Cura / Ideamaker i did not have this issue so should noting wrong with the printer itself.

#### Project File (.3MF) where problem occurs
Just created a cube without top layers. 2 perimeters, 5 bottom solid layers.

[image]
[image]
[image]
[image]
[image]
[image]

Lt-Kammello, 2023-01-22:
That’s not SS, it’s a print artifact due to 3 main cauese, excessive speed, excessive PA, probably too low flow rate for speed /  hotend temp, and a bunch of more tuning to doo.
Start going slower and decrease pa while observing print.
40mm/s should be fine, if you think that’s slow start from there anyway then maybe do some more test.

itchy2, 2023-01-23:
Hi Lt,

thanks for your reply. currently I print the Infill with 80 mm/s, so I will try to reduce to 40mm/s and try again. PA is not enabled on that prints. Too low flow rate was also in my mind but I could not find any option to increase just infill flow as external perimters and top infill are fine.

Thanks and kind regards
Itchy2

Lt-Kammello, 2023-01-23:
I'd suggest you to let SS do it's mathe.
Just set correct nozzle width under print settings, then set width and flow as follows:
[image]

This way youll let SS do its math wich is correct from my experience.
Than as we said start conservative on print speed, i use volumetric and find it fine.
PA impact really depend a lot on the proportion between speed and acc/dec.
Generally the higher the speed the lower acc/dec the more PA you need, but depends on many other variables, vice versa you generally need less of it.
I'd suggest you to give it a chance and calibrate it.
Follow the Andrew Ellis guid ytou can finde online by googling it, it's really escellent for starting up correctly.

Best luck =)

itchy2, 2023-01-23:
Hi,

thanks for your feedback. 

I tried the reduced speed approach. The Situation has improved but the infill is still not correct connected (i can crack it without big effort). I thought it is/was a superslicer issue as I printed the infill from cura / ideamaker without issues at 80mm/s.

Shall I only but with and flow back to "standard"/your values? shall I also bring speed back to standard?

Thanks and kind regards
Itchy2

itchy2, 2023-01-23:
short update: Just completed a print with all width and flow on default. Same issue. what I don't understand: the issue is always into one direction. so it is not a random issue. it's always the same side that is not connected.

neophyl, 2023-01-24:
Maybe I missed it but what infill pattern are you using ? Some infills cross over each other causing a 'bump' and others dont.  For something like cubic which is my usual it doesnt and prints a continuous line and I never get anything like that.  Grid is one of the crossing patterns.

itchy2, 2023-01-24:
Hi @neophyl 

it is the GRID-pattern. Do you mean that by crossing the line the printhead will be liftet a little bit and that causes the gap? That might be an interesting approach. but in CUBIC the lines are also crossing within one layer. I will try it with "lines" as there is not crossing within the same layer.

itchy2, 2023-01-24:
just completed my try with "lines" @ 40mm/s. result is not really satisfying. 

What I dont understand is why this issue never occured in cura / ideamaker.

[image]

Lt-Kammello, 2023-01-29:
That's a good result with Lines, since they are made at alternate layers and are not compacted like for instance Grid.
That's a good result with lines, this kind of infill doesnt make up for strong parts, it's supposed to be used for reducing material use one ahestetics parts.
You absolutely have no issue there, you might just want to teak PA a Bit.
The fact that defects are happening in a direction only it's a drage effect of the nozzle over the molten filaments.
Don't keep opening new posts man, those guys have a ton of work to do already.
Here's my discord tag, pm me if you need an help Lt. Kammello#4682

itchy2, 2023-01-30:
Hi Kammello, many thanks for your offer to support. I am happy to take it and I will close this topic here.

closed 2023-01-30 by itchy2