supermerill/SuperSlicer #3594: "Better bonding" setting is essential

ivizot, 2023-01-23:
A new setting called "better bonding" is found under "External perimeter first" section.

We have tried hundreds of settings combination and found out that this particular one is the only which is able to produce not just water-tight, but pressure-tight prints. Enabling this option allowed us to print "vessels" which hold up to several tens of bars of pressure with 4 perimeters only on 0.5mm extrusion width (PETg) and gyroid infill (15%). All other settings didn't give nearly as close result even at enormous over-extrusion (up to 150%). All parts were water-tight at most, and didn't hold even 1.5 bars.

So, I am kindly asking to save this feature for next releases and, if possible, to introduce such a feature as a stand-alone in perimeter generator. Also would be really helpful if such a feature is also available for solid infills.

inspectionsbybob, 2023-01-26:
Is this in Cura, Prusa, Simple Slicer or who?

Curious

ivizot, 2023-01-27:
Well, it's in SuperSlicer. I thought this is the issue tracker for it :)

Lt-Kammello, 2023-01-29:
A perfectly tuned print should have no gaps except under support or when all seams aligns.
 FDM is not the right technology for pressure vesserls etc, it can be done but that's not the right tool.
Tuning anything different from SS math rseults in either small or big defects, and "better bonding" basically force higher overlaps creating other artifacts, the utility is marginal and i think it creates more confusion and issues than it solves, should at least stay on "advanced" or "not recommended"