supermerill/SuperSlicer #3598: Holes in infill in some configurations

CODeRUS, 2023-01-23:
### What happened?

Hi there, I'm getting weird holes in solid infill and internal bridge.
Using "Ensure vertical thickness" is NOT a proper solution :)

Using nightly build from 21 Jan, also reproducible with older versions.

First screenshot: SuperSlicer 
[image]

Second screenshot: PrusaSlicer
[image]

Project file:
[hemera xs volcano - Fan mount plate flat.zip](https://github.com/supermerill/SuperSlicer/files/10484504/hemera.xs.volcano.-.Fan.mount.plate.flat.zip)

### Project file & How to reproduce

Use project file to reproduce.

### Version

2.5.59.2

### Operating system

Win10

### Printer model

n/a

CODeRUS, 2023-01-24:
@neophyl do yo mean this expected to give me following result:

[image]

If so then layer below now looks like this:

[image]

And this is totally unexpected behaviour :D

neophyl, 2023-01-25:
Is that the same part/project you uploaded ?  Cause I'm not seeing anything like that.  
[image]

and the layer below 
[image]

btw I had to change the minimum feature size in your project for arachne from 25mm to something more sensible as when originally opened most of the object when sliced was missing.  Dont know why it comes through as 25mm.  So I'm wondering if there's some profile loading errors/weirdness going on with these 2.5.xx releases which are not really proper releases anyway.  

If the setting doesn't work for you globally then you could just use a height range modifier to apply it at selected heights.

If not and you are getting a better result in a different slicer then just use that slicer.

CODeRUS, 2023-01-25:
@neophyl should be some profile export error, interpreting 25% as mm on export or import...

Thanks, i'll try to import my own project and check what is was missing or automatically optimized by that :D