supermerill/SuperSlicer #3602: Extruder "Reset to Filament Color" option doesn't properly work in Gcode preview screen

HitLuca, 2023-01-25:
### What happened?

Starting from a multi-extruder printer profile, where each extruder doesn't have an associated color, I tried to slice a multi-color object. 
The 3D view properly shows the object having different colors based on the associated filament.
The Sliced preview again shows proper coloring
The Gcode preview instead has the object shown entirely in black, and each extruder has an assigned black color.

If I manually change the extruder colors in the `Printer Settings/Extruder #/Extruder Color`, the Gcode preview works as expected, with each extruder showing its set color. The extruder colors override the colors in the 3D view and Sliced preview tabs as well, but that is to be expected

My colored object, as seen from the 3D view
[image]

What happens when `Extruder Color` is undefined (`Reset to Filament Color` clicked) -> Gcode preview WRONG, 3D view as before
[image]

What happens when `Extruder Color` is set to some color for each extruder -> Gcode preview OK, 3D view uses extruder colors instead of filament colors
[image]

### Project file & How to reproduce

Tested with the default `Prusa i3 MK3S & MK3S+ MMU2S` profile

### Version

2.4.58.5

### Operating system

Windows 10

### Printer model

Default profile -> Original Prusa i3 MK3S & MK3S+ MMU2S

barsamgr, 2024-02-15:
This is still an issue.  So instead opening a duplicate issue, I want to confirm this issue as of lates release.  By the way this issue does not exist in Prusa Slicer.

Sycha, 2024-07-24:
I am having the same issue.

Version 2.5.59.12