supermerill/SuperSlicer #3605: Z-Hop on Travel

inspectionsbybob, 2023-01-26:
### Version
2.4.58.8

### Operating system type + version
Win10

### Behavior
The Nozzle is prone to clipping supports durring travel moves WITHIN A LAYER.
Would Like to see a new Feature "Hop on Travel"  that would raise the Nozzle before Traveling.

Current HOP seems to be tied to retraction and layers, I would like to see one tied to TRAVEL

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
Almost any project file with Tall-Narrow supports, the supports are OFTEN clipped durring travel moves accross the support.
Would like to see "AVOID SUPPORTS WHEN TRAVELING" fixed and -ADD- a Z-Hop to it as well.  Don't want Z-Hop Everywhere as it really slows things down, but want something that will reduce the likelyhood of a support being clipped.

I can sometimes HEAR it crossing the InFill  (Brrrrrr) .  Not a problem, but it can't be good for the Nozzle, and could PROBABLY help cause layer shifts

If there is a way to impliment this NOW using current features that won't inject lots of unneeded Hops at every retraction, please describe how to impliment it.

Garryb123, 2023-01-26:
Great feature request. I experienced that issue yesterday.

Because we may not want SuperSlicer to Z-hop always within a layer (that could end up being a lot of hops depending on the part) I'd suggest 'Z-hop over support' and 'Z-hop over support AND within layers' in addition to regular Z-hop. This should take care of all cases?

inspectionsbybob, 2023-01-26:
I would be happy if "avoid supports" was enforced.  The files I sent showed that there were several routes right through the support, and others that went to/from the middle... Nothing should be going THROUGH the support location, and it should HOP to/from the middle to avoid problems.