supermerill/SuperSlicer #3606: mm/s speed overrides in filament

guestisp, 2023-01-27:
Currently, there is just one volumetric speed override in filament tab.
Having the speed in mm/sec2 like in the standard Speed page would be better as you can set the same speed you have in the other page.

Also, being able to override all speeds would be even more useful.

Lt-Kammello, 2023-01-29:
You can't because speed is not volume.
Different lines have different width in order to account for filling gaps between them, that's how SS math works.
If you set Volumetric it doeas mean that speed sligty changes in order to keep the volume, if you set a fixed speed thi does not happen.
Just usa a volumetric calc you can find online, there's many.

guestisp, 2023-01-29:
but why on the speed page we have mm/s and not the vioumetric?

guestisp, 2023-01-30:
the exact setting i'm asking for is also available in the height range modifier, so why isn't possible to override the speed in mm/s in the filament page ?