supermerill/SuperSlicer #3609: Single Extruder MM Cannot Get Filament/Extruder Pairing to Save??

dohlin, 2023-01-28:
### Version
2.4.58.5

### Operating system type + version
Windows 11 (Latest)

### Behavior
SuperSlicer with single extruder multmaterial enabled won't save my latest extruder/filament choices - it constantly reverts to all showing "Generic ABS" and requires resetting.  It _seems_ to happen most often when I change which physical printer I'm using, but I am pretty sure I've seen it do it other times too and it's driving me nearly insane :)

#### Project File (.3MF) where problem occurs
N/A

What things should save like:
[image]

What things revert to constantly instead:
[image]

kursatu, 2023-02-12:
It has been working for me for a long time. It  would have been a major hassle.

xF4m3, 2024-03-04:
> It has been working for me for a long time. It would have been a major hassle.

For me it does save these Filament allocations, but only when i use one printer.
As soon as i switch to a diffrent printer that does not have the MM setup or a diffrent one, the Filament and Extruder mappings revert to the First Filament i have in the list on all of the Extruders.

Has really been bugging me alot to always fix the mappings.

neophyl, 2024-03-04:
Yes it will revert when you change printers.  Different printers support different things so it resets everything.

Take advantage of saved projects.  Configure them how you want them to be (common colour extruder setups etc for each printer) and then load them in when needed.  You can use them as libraries of favourite settings as well as simply model slices.

xF4m3, 2024-03-10:
Yes that is one way of getting around this issue, a band aid, but really this should be information that gets saved per printer.

Making multiple project for diffrent setups just makes it a hassle to keep on top of them when you tune other printer settings which you have to also save for every one of those profiles. Would love to see this bug fixed, i do not think it should be that much work to save this additional info, though i am no programmer so i might be wrong about that.