supermerill/SuperSlicer #3613: Wall thickness dont go under 0,8 (Nozzle 0,4)

Jogiarea, 2023-01-30:
### What happened?

Hello,

have a problem, I can not print thin walls, the setting Wall thickness can not be put down.
I use an ender 5 and Voron V0 with 0.4mm nozzle each.

If I set to below 0.8mm it automatically corrects upwards.

The result is that when I print only one wall, it is printed with a line at 0.8, but I want only 0.4 thick walls.

Is this a bug or am I missing something?

Happens to me in the release and the current nightly

Tested with Prusaslicer, there it prints as wanted 0,4

Translated with www.DeepL.com/Translator (free version)

### Project file & How to reproduce

Kant Print thin walls

### Version

2.4.58.5

### Operating system

Windows 11

### Printer model

Voron V0 Ender 5

DKS-18, 2023-01-30:
can you share your project file?

Jogiarea, 2023-01-30:
> can you share your project file?

Sorry, forgotten to upload it see first comment

Jogiarea, 2023-01-30:
OK, same Problem in Prusaslicer, i tried to print this file with wallthicknes of 0,45 and the printer prints one with a wall, a wall thickness of 0.8mm
I do not find anything to change to end this problem
[Zahnbürstenhalter_Dieter Geli v2.zip](https://github.com/supermerill/SuperSlicer/files/10537282/Zahnburstenhalter_Dieter.Geli.v2.zip)

Prusaslicer shows this
[image]

Will still test a cylinder without cap with no infill to print to see what comes out there

DKS-18, 2023-01-30:
do you have i thin wall enabled?
[image]

Jogiarea, 2023-01-31:
hi yes have thin walls activated, actually always with all slicers, also the values are under 100%,
I'll turn it off and do another test print tonight

vespaman, 2024-12-18:
Hi, 
I think I have the same problem; with 0.6 nozzle, my wall will not go under 1.2mm (I want 0.8mm).  These walls are also perimeters, so I tried to reduce the setting above "thin walls", named "Overlapping external perimeter" which says it can disable thin walls on external perimeters. I can confirm that I do not get any thin walls according to the "Filament/Feature types" view in "sliced view", so I suspect this value (80%) is what is causing the issues.

But:- reducing the "Overlapping external perimeter" to 0, -1 or indeed any value below 71% just makes SuperSlicer crash.

Is there a way around this?

I'm on Version 2.5.60.0