supermerill/SuperSlicer #3614: Extrusion width percentage calculation based on layer height

Henk72, 2023-01-30:
**Is your feature request related to a problem? Please describe.**
SuperSlicer (SS) allows to set extrusion width as a percentage of the nozzle width. PrusaSlicer sets the percentage based on the layer height. The latter is useful in case one uses variable layer height (VLH). This functionality is also available in SS and can change the layer height.

**Describe the solution you'd like**
A checkbox to determine if the extrusion percentage should be based on nozzle width or layer height.

**Describe how it would work**
With the checkbox the system bases the extrusion width calculation (if a percentage is defined) either on  the nozzle width (default) or on the actual layer height. 
With this option the extrusion width can follow the layer height in a defined ratio. It is set in the same way as it is done now by setting the extrusion width percentage. 
Example:
Nozzle 0.6
Variable layer height is between 0.16 and 0.38mm.
Extrusion width is set to 300% and the new checkbox is set to layer height.
The extrusion width at layer height 0.16 would be 3 * 0.16 = 0.48
The extrusion width at layer height 0.38 would be 3 * 0.38 = 1.14

**Describe alternatives you've considered**
Manually changing extrusion width with layer changes set manually. Very cumbersome.

**Additional context**
With the percentage one can maintain a fixed extrusion width to layer height ration in the whole print that might have changing layer heights. This is very helpful if one wants to have a strong layer bonding (ratio > 3).

Dx-Dx-Dx, 2023-02-01:
Why? How is it useful? I never understood it in PrusaSlicer. Why extrusion width must change with layer height? What does it do?

Henk72, 2023-02-01:
It is useful if one makes a functional part that requires high strength. One of the factors to achieve high strength is to have good layer adhesion (and fill). This can be realised by printing wider than higher (a flat rectangle instead of a square). A height to width ratio of 1:4 is a good value. 

If one prints a model that needs lower layer heights at certain places, but higher for vertical areas, SS and PS supports this with variable layer height. The layer height is set (and can be changed by the user) before slicing. With my checkbox suggestion, the width can be calculated (for example using this 1:4 ratio).

For background of the ratio, I like to refer to this research article about print strength (just scroll to the x-ray picture, it explains it graphically): https://www.researchgate.net/publication/323572560_Strength_of_PLA_Components_Fabricated_with_Fused_Deposition_Technology_Using_a_Desktop_3D_Printer_as_a_Function_of_Geometrical_Parameters_of_the_Process

kursatu, 2023-02-12:
I generally agree that this is useful. 
You can implement this with a post processor while waiting for the feature to be implemented. 
I have implemented a rather complex post-processor that addresses the shortcomings of both SS and PS. 
It deals with wipetower speed/insertion, support interface thickness for MMU and e-acceleration improvements.
Eg. the support interface processor changes the extrusion by a ratio of max_layer_height/layer_height.
That way the overhang support interface layers are printed rather thick, which allows very good separation of the support material. Also , small layer height causes very little material to be extruded which is is very hard to print  because there is usually no layer underneath.
What you are suggesting requires almost this same adjustment.

kursatu, 2023-02-12:
I do print functional pieces and I might give it a try at some point , when/if it makes it into the TODO list.

Henk72, 2023-02-13:
Kursatu, nice idea about your post-processor. Could you share your code? I saw a German word in your comments, I can read German if needed.

For the accelerations I used SS which can set the acceleration per extrusion type with M204 commands. I changed this macro in Klipper to also set travel acceleration and square corner velocity.

Looking forward to your post-processing!