supermerill/SuperSlicer #3616: Can't install 2.4.58.5 on my Intel Mac with Os Ventura.

Jonibygg, 2023-01-31:
### Version
2.4.58.5 for Intel
Filename: SuperSlicer_2.4.58.5_macos_220924.dmg

### Operating system type + version
Mac OS Ventura 13.2

### Behavior
I download the release version of SuperSlicer from GitHub an try to run it but end up with this message

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: <EA3F7209-F390-3268-9C90-D8081DCEDC30> /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

jimmyD30, 2023-02-03:
Try 2.5.59.0, it's working for me on same OS, but if I remember correctly I had to replace (or simply add) the popup image in the app bundle, it didn't cause a crash, just an error message - if you want more info reply back.

Jonibygg, 2023-02-03:
Thank you, that worked after I found it.
I’m not that comfortable with GitHub I can say.