supermerill/SuperSlicer #3625: What did I turn On?

inspectionsbybob, 2023-02-03:
### Version
2.4.58.5

### Operating system type + version
Win10

### Behavior
What did I turn on and How do I turn it Off?
All of a sudden I am getting warning about obects too close to each other 
or
Too Tall (there are not too tall)

What did I turn on by mistake, and how do I turn it off?

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
[Weird errors.zip](https://github.com/supermerill/SuperSlicer/files/10582952/Weird.errors.zip)

herrtschaefer, 2023-02-04:
In the slicer there is an option to print objects sequentially, so it completely prints one then the next. In the slicer there are options to define the tool clearances so the tool does not collide with the finnisches parts.

inspectionsbybob, 2023-02-04:
What is the title so I can do a search on it please?

I thought it was that but couldn't remember the right word to search for...

herrtschaefer, 2023-02-04:
Go to Print Settings->Output options and untick the checkbox Complete individual objects.