supermerill/SuperSlicer #3626: Fix the Object Manipulation Screen Size & Locations

inspectionsbybob, 2023-02-03:
### Version
2.4.58.5

### Operating system type + version
Win10

### Behavior
The Object Coordinates for -Z- are really messed up.

The Object on the screen may be 50.22mm tall, but it is NOT 25.11mm off the build plate.

If I click on "Drop to Bed" the -Z- position should go to ZERO as it is ON THE PLATE.

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
[Z-error.zip](https://github.com/supermerill/SuperSlicer/files/10583039/Z-error.zip)

neophyl, 2023-02-05:
Object coordinates are done from their centre Bob.  This is how PS does it and there have been a fair few 'issues' raised relating to the whole co-ordinate system over on their github too.

inspectionsbybob, 2023-02-05:
I understand that for x & y.

but Z makes no sense, it is not from the middle, that would be HALF of of its height and if you zero'd the z-height it would sink the model 1/2 into the bed.

neophyl, 2023-02-05:
Yes, thats exactly what it does.  If I add a part that is 31.22mm high its centre is 15.61mm above the bed.  If I look in the right hand panel the Z height is listed as that.
[image]

If I type in zero to the Z position then it sinks the part half way into the plater. 
[image]

This to me makes perfect sense.  Its logical and consistent with X/Y too.  
This is why there's is the place to bed control there as that makes it easy to to that job without the user having to work out any math.  We all know users don't like to work that out don't we.

inspectionsbybob, 2023-02-05:
Ok, I will go with that.

I will revist some models where it didn't seem to work that way...

but close for now...

inspectionsbybob, 2023-02-09:
Ok, found on of the models that makes no sense.
The two objects are the same, the one on the right was set to ZERO z-height, and it sank it less than 1/2 the height...

The One on the left showed the Height at 94 hight, and an 11.x offset...

So from those numbers, what is the actual size?
[image]

neophyl, 2023-02-09:
Come on Bob, attach the project :)  lol  I hate having to try and guess from screen caps.

inspectionsbybob, 2023-02-09:
[camera_mount_v2.zip](https://github.com/supermerill/SuperSlicer/files/10700580/camera_mount_v2.zip)

Sorry, thought this was kind of simple

inspectionsbybob, 2023-02-09:
It was a STL from thingiverse...
Why when I load it clean from thiniverse does it get the height wrong?

neophyl, 2023-02-09:
Are you sure it’s a stl ?when you load in an stl it uses the type as part of the name in the tree so xxxxx.stl. 
Those in your project don’t have stl as any part of the name. Also when it’s saved as an stl from slicer and then loaded it behaves as you would expect an stl to behave.

So again I’d say it’s not a stl file.  Stl are really dumb, they don’t contain much besides the coordinate data. Other types can contain all sorts of extra data. Without the original file you used then it’s impossible to say.