supermerill/SuperSlicer #3630: Crash spiral vase selection 2.5.59.2

dentinox, 2023-02-04:
### What happened?

When doing a calibration cube https://cdn.help.prusa3d.com/wp-content/uploads/2021/04/cube-40-40-40.zip and you select spiral vase, superslicer asks to adjust some settings. When you click yes it crashes back to windows.

settings 0.32
per 2
Wall thickness 0.8
Layers 4

### Project file & How to reproduce

Print settings
select spiral vase
Dialogbox YES.

### Version

2.5.59.2

### Operating system

Windows 10

### Printer model

Ender 5

yuri-rage, 2023-02-10:
Identical behavior here with 2.5.59.2 nightly. Easily reproducible (crashes every time vase mode is attempted).

JacTheMaker, 2023-02-11:
Same here Windows 11
On multiple print and filament settings.

cypeks, 2023-02-25:
Same here 2.5.59.2 nightly build GTK3 - Ubuntu 22.10

kymlalu, 2023-03-04:
Same here - Win10. 
For me last version where vase mod didn't crash SS was/is 2.5.59.0. Everything after just crashing.

cypeks, 2023-03-06:
Last 2.5.59-2 pre-release also crashing.

Lemcott, 2023-03-07:
same - WIn11 Ryzen - SuperSlicer_2.5.59.2_win64_230304

liftbag, 2023-03-07:
Latest mac version crashes too.

shadowdro, 2023-04-05:
Same issue here.

Fropzs, 2023-04-15:
How to fix it? Same crash here.

artemen, 2023-05-28:
one more here

EvilGremlin, 2023-08-09:
do not exist in https://github.com/supermerill/SuperSlicer/actions/runs/5548343816

EvilGremlin, 2023-09-20:
its already fixed, update more

jaitti, 2023-11-06:
Just started with superslicer and noticed that both win and linux versions crash when trying to activate spiral mode. There is a message here pointing to a newer build which does not have any actual builds. All artifacts has been expired. It looks like it has been broken for 9 months now.. Is this slicer dying?

closed 2024-02-11 by supermerill