supermerill/SuperSlicer #3631: Temperature

ljungis, 2023-02-05:
### Discussed in https://github.com/supermerill/SuperSlicer/discussions/3627

<div type='discussions-op-text'>

<sup>Originally posted by **ljungis** February  4, 2023</sup>
[Body.zip](https://github.com/supermerill/SuperSlicer/files/10608238/Body.zip)

In Filament Settings the temperatures are:  Extruder      First layer 250 degr.                  Other layers 245 degr.
                                                                     Bed             First layer 60 degr.                     Other layers 50 degr.

However when printing only the bed temperature changes, the extruder temperature doesn't, it stays at first layer setting 250 degr.</div>

neophyl, 2023-02-05:
I presume you generated that project in 2.4 ?  As when I slice it in 2.5.59 it gives me an access violation and shuts down.  When opened in 2.4 and sliced it works ok.
Here is the thing though, when sliced in 2.4.58.4 it slices with the first layer at 250 and the later layers at 245.  Which is exactly what is in your filament profile.  So I cant see the issue.
[image]

If I save the gcode then at the layer change we have the following (starting at line 1398)
;LAYER_CHANGE
;Z:0.47
;HEIGHT:0.22
; stop printing object Huv-Body.stl id:0 copy 2
M486 S-1
G1 Z0.37 F7800
**G10 P0 S245 ; set temperature**
M140 S50 ; set bed temperature
G92 E0
G1 E-1.2 F1200
G1 Z0.57 F7800
; printing object Huv-Body.stl id:0 copy 0
M486 S0

You can quite clearly see the set temperature command in there.  The 'problem' is your start gcode block is using the M104 command to preheat though and the sliced print command is using G10.  Looking at the reprap info G10 was/is the original extruder temperature command.  So G10 P0 S250 is a valid command for Reprap based printers to tell it to set Tool 0 to 250 degrees. 
I do see that they have updated newer versions of reprap to also use the same M104 that Marlin uses but the reprap 'flavour' of SuSi supports the original standard.  Changing it would probably break compatibility with older printers and stop them working.
So it looks like you need to choose a more suitable firmware flavour as if your printer was using Reprap then it should accept both the G10 and the M104 as valid commands.  Which it obviously is not doing.  

If you aren't familiar with where that is set its in Printer Settings>General>Firmware>G-Code flavour.

ljungis, 2023-02-05:
Thank you neophyl. It is correct that I am on vers. 2.4.58.5.
But how come it works for bed temperature?

ljungis, 2023-02-05:
I have now set G-code flavor to Repetier. The temperature settings works fine. Thanks!

closed 2023-02-05 by ljungis