supermerill/SuperSlicer #3635: Mac OSX: Crash when launching application (SuperSlicer_2.4.58.5_2022-09-23)

rr12267, 2023-02-08:
### What happened?

Downloaded asset
copied to applications folder
Ctrl click to open
.
.
.
Crash occurs (looks like libzstd is not included in your build package.)

Crash Report:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [1339]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-02-07 18:48:18.6265 -0700
OS Version:            macOS 12.6.3 (21G419)
Report Version:        12
Anonymous UUID:        CDF38DE7-C864-473E-F4A9-639A5642BC1B

Sleep/Wake UUID:       379BA59E-D330-4353-9673-C9AD4CAF961A

Time Awake Since Boot: 5500 seconds
Time Since Wake:       1366 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x114bdb98e __abort_with_payload + 10
1   dyld                          	       0x114bf1a72 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x114bf1aa4 abort_with_payload + 9
3   dyld                          	       0x114b9f699 dyld4::halt(char const*) + 615
4   dyld                          	       0x114b9b5b5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x114b9a4e4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b435d628  rdx: 0x00007ff7b435da90
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b435d670  rsp: 0x00007ff7b435d628
   r8: 0x00007ff7b435d690   r9: 0x0000000000000000  r10: 0x0000000000000075  r11: 0x0000000000000246
  r12: 0x0000000000000075  r13: 0x00007ff7b435da90  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x0000000114bdb98e  rfl: 0x0000000000000246  cr2: 0x0000000114ba1da4
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x114b95000 -        0x114c00fff dyld (*) <006a3e6f-3cd3-34d9-b0f2-ed6bd67a95a6> /usr/lib/dyld

[... 112 lines cut ...]
    }
  ],
  "experiments" : [

  ]
}
}

### Project file & How to reproduce

1. Download asset [SuperSlicer_2.4.58.5_macos_220924.dmg](https://github.com/supermerill/SuperSlicer/releases/download/2.4.58.5/SuperSlicer_2.4.58.5_macos_220924.dmg)
2. copy to applications folder
3. Ctrl click to open as getting cannot verify developer message.

### Version

2.4.58.5

### Operating system

mac os x 12.6.3

### Printer model

ender 3 pro

rr12267, 2023-02-08:
I needed to install zstd library via brew (brew install zstd) for the application to launch.

Morcegolas, 2023-02-16:
> brew install zstd

That resolve my problem.
Thanks.

drpeej, 2023-08-07:
I am still getting this problem