supermerill/SuperSlicer #3636: Using Height range modifier to change #perimiters does not work on final layer

EmanuelStassar, 2023-02-09:
### Version
Version 2.5.59
### Operating system type + version
Windows 10 Pro 21H2

### Behavior
* _Describe the problem_ I want to print a single wall airplane wing, but the first and last few layers I want 2 perimeters. I am using height modifiers to do this. This print has 0 top and bottom layers.
* _Steps needed to reproduce the problem_
* _Expected Results_ I expected 2 perimiters on the sections I specify, but that does not work for the top layer. At the bottom it works as expected
* _Actual Results_ The final layer only has one perimeter.
  * _Screenshots if useful_ Added 2 screenshots, one of the 3D model, one of the sliced result.
[image]
[image]
_Is this a new feature request?_ No, maybe a bugfix.
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
[Wing-1_2.zip](https://github.com/supermerill/SuperSlicer/files/10696546/Wing-1_2.zip)

_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

neophyl, 2023-02-09:
You do realise that your print settings profile has 'only one perimeter on top surfaces'  enabled ?  Can you guess what happens when you untick that setting ?  ;)

EmanuelStassar, 2023-02-09:
Oooh, I must have overlooked that one. Sorry for the trouble and thanks for your help! This slicer has soo many options that its easy to miss stuff like this.

closed 2023-02-09 by EmanuelStassar