supermerill/SuperSlicer #3638: Max detour length won't accept Percentages

inspectionsbybob, 2023-02-09:
### Version
2.4.58.5

### Operating system type + version
Win 10

### Behavior
I tried to put 150% (50% more than the direct path) into "Max detour Length," and it will only take NUMBERs (50-150 etc.) not 150%  Actually, when I just tried to confirm this, it won't let me enter anything but "0"

I think the entire "Avoid Crossing..." area needs some help.  I saw someone -else- opened a ticket about crossing perimeters and compared it to Prusa with some nice comparison images.   

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Proj
[camera_mount_v2.zip](https://github.com/supermerill/SuperSlicer/files/10700638/camera_mount_v2.zip)
ect as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

neophyl, 2023-02-09:
I just tried that setting in both 2.4.58.5 and 2.5.59 and I can enter percentages just fine.
[image]

inspectionsbybob, 2023-02-09:
Ok, so what is blocking, all it would let me enter was -0-

neophyl, 2023-02-09:
No idea what’s blocking it. I even opened and used your other project with the z position problem and I could set it that too. 

Given the number of these types of issues you have I have to wonder if there’s something broken in the profiles you are using.