supermerill/SuperSlicer #3642: Snug Supports not being generated (late 2.4) (2.4.58.5)

AEgorovCCR, 2023-02-11:
### What happened?

Jnly bottom and interface layers of supports generated under thin geometry.
[image]
[image]

### Project file & How to reproduce

See my 3mf file in late 2.4 (2.4.58.5) version
[bad_supports.zip](https://github.com/supermerill/SuperSlicer/files/10713517/bad_supports.zip)

### Version

2.4.58.5

### Operating system

Windows 10 x64

### Printer model

Custom CoreXY

Pavulon87, 2023-02-11:
For a quick workaround, try changing the "Support layer height"

AEgorovCCR, 2023-02-11:
> For a quick workaround, try changing the "Support layer height"

2.4 dies not have that setting. Or I cant find it. I know its in 2.5. But 2.5 gives me missing layers in supports.

Benjamin4456, 2023-03-05:
Another work-around I discovered is to reduce the x-y separation distance.  It's not a perfect fix, but it can be helpful on occasion.  Of course it can also introduce its own issues, so be sure to check that your supports aren't fusing to the walls of your parts.