supermerill/SuperSlicer #3643: Segfault when slicing with avoid crossing perimeters on

kozross, 2023-02-12:
### What happened?

When I slice a file with 'avoid crossing perimeters' on, SuperSlicer segfaults. Specifically, the error given is this (when run from a terminal):

```
superslicer: /usr/src/debug/superslicer/SuperSlicer/src/libslic3r/GCode/AvoidCrossingPerimeters.cpp:405: Slic3r::Direction Slic3r::get_shortest_direction(const AvoidCrossingPerimeters::Boundary&, const Intersection&, const Intersection&, float): Assertion `dist_second >= 0.f && dist_second <= contour_length' failed.
Aborted (core dumped)
```

However, if I disable that setting, the file slices just fine. The error seems to occur on layer 121 based on what I saw.

### Project file & How to reproduce

File to reproduce with: [even-more-ridiculous-speed.zip](https://github.com/supermerill/SuperSlicer/files/10715594/even-more-ridiculous-speed.zip)

1. Open SuperSlicer
2. Load the attached file
3. Try to slice
4. Watch the segfault in real time

### Version

2.4.58.5

### Operating system

Arch Linux

### Printer model

Neptune 3 Pro

Pavulon87, 2023-02-13:
Tested 2.4.58.5(and 2.5.59) on Windows 10 and Ubuntu 22.04 - no problems at all.
Maybe not enough RAM?

kozross, 2023-02-13:
@Pavulon87 - the computer in question has 32GiB of RAM.