supermerill/SuperSlicer #3644: SuperSlicer makes bridges mid air - would be nice to resolve.

RiccaDS, 2023-02-13:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
[image]
The image speaks for itself. When there is a surface to be obtained with bridges, and in this surface there is, for example, a hole, what I experienced is that SuperSlicer goes and tries to make the hole borders before. Of course, as it tries to make it mid-air, it turns out all but a circle. 
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
It would be very nice if it would be possible to find a path strategy for this kind of geometry without supports. 

**Describe how it would work**
Be as descriptive as your skill level allows you.
I can imagine SuperSlicer starting to bridge for example diagonally where the bridge distance is smaller, then making it's way to longer paths with the side sustain of the shorter bridge lines previously printed. When it encounters one of the holes it slides on the adjacent bridge line and clean the nozzle so as to avoid collapse of the bridge. Only when the entire surface is made, the circular borders of the hole can be printed. At least theoretically.  Of course it could be used also with other geometries.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Supports, but I've not been able to use supports with good results in PrusaSlicer. I haven't tried in SuperSlicer yet. But in the case, very small contact points between model and supports would be much better.

**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2023-02-13:
There is a strategy already.  Its called fill in the hole for a single layer to make a uninterrupted bridge.  Normally you would design it into your model but SuSi already has this option available.   Then when printed its very easy to poke out the single layer to give you the hole.

Print Settings>Perimeter & Shell>Advanced>No Perimeters on Bridge areas.  There are several options available.  Experiment and look at what the various options do.

RiccaDS, 2023-02-14:
ok thank you, I am trying to make it work, still not successefully however . It doesn't seem to be wanting to fill the hole. But I am trying further, I am trying to tweak the options in some way.