supermerill/SuperSlicer #3655: Super slicer doesn't open on Mac, just crashes

robmc7683, 2023-02-16:
### What happened?

The application crashes before it opens.

I have tried the latest 2.4 edition from September 23, 2022

As soon as you try to open or install the program it crashes.

This is the report I get.
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [66223]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-02-16 13:01:39.2667 -0600
OS Version:            macOS 12.6.3 (21G419)
Report Version:        12
Anonymous UUID:        D47A2F83-AB55-76AA-D507-2A2DC712F5A3

Sleep/Wake UUID:       968D5BCC-2E14-45AF-A068-86C6D9DC1C80

Time Awake Since Boot: 38000 seconds
Time Since Wake:       713 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x10dc4e98e __abort_with_payload + 10
1   dyld                          	       0x10dc64a72 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x10dc64aa4 abort_with_payload + 9
3   dyld                          	       0x10dc12699 dyld4::halt(char const*) + 615
4   dyld                          	       0x10dc0e5b5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x10dc0d4e4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b9f5b528  rdx: 0x00007ff7b9f5b990
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b9f5b570  rsp: 0x00007ff7b9f5b528
   r8: 0x00007ff7b9f5b590   r9: 0x0000000000000000  r10: 0x00000000000000d8  r11: 0x0000000000000246
  r12: 0x00000000000000d8  r13: 0x00007ff7b9f5b990  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x000000010dc4e98e  rfl: 0x0000000000000246  cr2: 0x000000010dc14da4
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x10dc08000 -        0x10dc73fff dyld (*) <006a3e6f-3cd3-34d9-b0f2-ed6bd67a95a6> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
[... 219 lines cut ...]
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Display: Color LCD, 2880 x 1800 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort:
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB30Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1

### Version

2.4

### Operating system

Mac OS 12.6.3

### Printer model

Creality ender 3 S1 Pro, Anycubic Vyper

Morcegolas, 2023-02-16:
I'm getting the same problem in my Mac running the same OS 12.6.3.
I'm compiling the software now to see if it helps.

Morcegolas, 2023-02-16:
brew install zstd

This resolved my problem.

Darkstar77, 2023-02-18:
This is your problem:

```
Termination Reason: Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)
```

You're missing a library, as @Morcegolas has pointed out. Install the missing library, and you should be good to go.

closed 2024-02-11 by supermerill