supermerill/SuperSlicer #3659: 756 Open Edge

raymadigan, 2023-02-19:
### What happened?

I have an stl file that was generated in freecad from two different bodies.  The bodies were combined using the lattice2 workbench and Attached Placement feature.  
[image]
If I look at each of the two objects individually, they both have zero errors.  

### Project file & How to reproduce

[IIRing.zip](https://github.com/supermerill/SuperSlicer/files/10777791/IIRing.zip)

### Version

2.4.58.5

### Operating system

Ubuntu 22.04.2 LTS

### Printer model

Ender 3 S1 Pro

Pavulon87, 2023-02-23:
What does this have to do with SuSi?
Blender also shows 756 non manifold edges.
[image]

raymadigan, 2023-02-24:
I attempted to close it after I submitted it but couldn't sort it out.

closed 2023-02-24 by raymadigan