supermerill/SuperSlicer #3680: Crash during slicing Voron parts

ilyapavkin, 2023-03-02:
### What happened?

Superslicer crashes during slicing. Last message in terminal:
```bash
superslicer: /usr/src/debug/superslicer/SuperSlicer/src/libslic3r/ExtrusionEntity.hpp:461: virtual const Slic3r::Point& Slic3r::ExtrusionLoop::last_point() const: Assertion `this->first_point() == this->paths.back().polyline.points.back()' failed.
[1]    101157 IOT instruction (core dumped)  superslicer
```

I use default profile for voron 2.4 with minor tweaks.

### Project file & How to reproduce

[front_skirt_a_350.zip](https://github.com/supermerill/SuperSlicer/files/10875061/front_skirt_a_350.zip)

### Version

2.4.58.5

### Operating system

Arch linux 6.2.1

### Printer model

voron trident

eduncan911, 2023-03-02:
I've been experiencing this ever since current version (Oct 2022)...  On discord, no one was able to help.

I've noticed that, most of the time (but not *all* the time), if you slightly move a single part, it slices.  That is all it takes, just move an item 1 or 2mm over.

I'm current fighting a few STLs right now that the "move" trick doesn't work.

ilyapavkin, 2023-03-03:
@eduncan911 that kind of worked! "Move" did not, as I initially encounter a problem within a batch print project, so I did both move and turn this model 90 deg. I just managed to find a possition for it - turn -1 deg on a Z axis (clockwise). That did not work other way though, nor of for turns less then 1 deg.

Anyway, that is a progress already. Despite 1 to 1 copy of profile back to PrusaSlicer, it reported to me double the estimated time. You just saved me a 7 hrs, thank you! :)

ilyapavkin, 2023-03-03:
Oh, and just to follow up - opposite side of that part, the `front_skirt_b_350` also causing same problem, however turns of +- 1 deg do not work. 89 deg turn got through.

eduncan911, 2023-03-03:
Hehe... cool!  Was just leaving another comment for @supermerill about this very thing.

eduncan911, 2023-03-07:
> Oh, and just to follow up - opposite side of that part, the `front_skirt_b_350` also causing same problem, however turns of +- 1 deg do not work. 89 deg turn got through.

Question: were you using the gtk2 AppImage?  I switched to the non-gtk2 appimage (i'm guessing gtk3) and no more crashes!

Then I upgraded to the latest 2.5, non-gtk2, and it continues to slice a 3mf that previously 100% failed.

Maybe you can switch appimages?

kubik369, 2023-04-09:
+1, exact same error, also have this happening in a similar timeframe to @eduncan911. I am on Archlinux, using the `superslicer-bin` from AUR. One part where I was able to get it to crash reliably is this keystone skirt from hartk [0] --- just import, rotate by 45° and it crashes at 0%.

[0] https://mods.vorondesign.com/detail/OVmVd5MNR1duzzLBA8zM2w

ilyapavkin, 2023-04-29:
I can confirm, that my parts now slicing properly as well. IMHO it is correct to close this issue for now, but feel free to reopen it if some odd model will cause this issue again.

closed 2023-04-29 by ilyapavkin