supermerill/SuperSlicer #3686: Superslicer crash when I activate vase mode

DUFUTUR, 2023-03-04:
### Version
2.5.59.2

### Operating system type + version
Windows 11 Pro (22H2) 64bit FR

### Behavior
Superslicer crash when I activate vase mode

[image]

#### Project File (.3MF) where problem occurs
[Forme-Box.zip](https://github.com/supermerill/SuperSlicer/files/10889565/Forme-Box.zip)

DUFUTUR, 2023-03-04:
I just complete the ticket with a capture of the evente log

[image]

mtaumike, 2023-03-04:
Same here, macOS on either ARM or Intel (don't use my Intel one much anymore).

Built from source via tag `2.5.59.2`

- `./BuildMacOS.sh -x`
- `./BuildMacOS.sh -a`

It is getting a segfault on both architectures - and I deleted the logs cause I was dumb.

WillCap397, 2023-03-06:
Same happens on Windows 10 64-bit

Creo2005, 2023-03-06:
The same on macOS

fr0sty1, 2023-03-10:
Linux Mint 19.1 as well:

```Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x0000555555b20788 in Slic3r::DynamicPrintConfig::normalize_fdm() ()
(gdb) where
#0  0x0000555555b20788 in Slic3r::DynamicPrintConfig::normalize_fdm() ()
#1  0x0000555555afae1f in Slic3r::Print::apply(Slic3r::Model const&, Slic3r::DynamicPrintConfig) ()
#2  0x000055555644803d in Slic3r::BackgroundSlicingProcess::apply(Slic3r::Model const&, Slic3r::DynamicPrintConfig const&, Slic3r::DynamicPrintConfig const*) ()
#3  0x00005555561a2fa7 in Slic3r::GUI::Plater::priv::update_background_process(bool, bool) ()
#4  0x00005555561a929c in Slic3r::GUI::Plater::priv::update_restart_background_process(bool, bool) ()
#5  0x000055555744e24e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
#6  0x000055555744e624 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
#7  0x000055555744e7af in wxEvtHandler::TryHereOnly(wxEvent&) ()
#8  0x000055555744e84a in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
#9  0x000055555744e8cd in wxEvtHandler::ProcessEvent(wxEvent&) ()
#10 0x000055555744f287 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
#11 0x000055555742b011 in wxTimerImpl::SendEvent() ()
#12 0x0000555557366948 in timeout_callback ()
#13 0x00007ffff5006e23 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff50063a5 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff5006770 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff5006a82 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff6784a25 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#18 0x000055555728af25 in wxGUIEventLoop::DoRun() ()
#19 0x00005555573a0a2d in wxEventLoopBase::Run() ()
#20 0x000055555736c14d in wxAppConsoleBase::OnRun() ()
#21 0x00005555573efa72 in wxEntry(int&, wchar_t**) ()
#22 0x000055555613c4f2 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#23 0x0000555555904c58 in Slic3r::CLI::run(int, char**) ()
#24 0x00005555558b7c7a in main ()
```

NovaViper, 2023-03-11:
Same issue occurs on EndeavourOS

paulgiordano, 2023-03-11:
So the problem is in src/libslic3r/PrintConfig.cpp line 8058:
`this->opt<ConfigOptionEnum<PerimeterGeneratorType>>("perimeter_generator", true)->value = PerimeterGeneratorType::Classic;`
Comment this line out and you won't get the segfault.
<p>I believe the code is attempting to enforce Classic (instead of Arachne) mode if you chose Spiral_Vase. The default, set in the top part of the code, is Classic, so I believe that unless you have specifically selected Arachne prior to this and then select Spiral_Vase you'd get some bad gcode, but that's just a guess on my part.
<p>I'm not a c++ person by any means, so I don't know the appropriate way to set the value here, but if there is a good c++ programmer out there who understands this code they could probably fix it easily.
As always, YMMV.

G-Pereira, 2023-04-25:
Same on Linux (Arch Linux)

WarlockSVG, 2023-05-25:
I just found out this happens when I want to use the vase mode.
Using Super slicer 2.5.59.2 on Windows 10 (64bit)

Gelisob, 2023-07-01:
> So the problem is in src/libslic3r/PrintConfig.cpp line 8058: `this->opt<ConfigOptionEnum<PerimeterGeneratorType>>("perimeter_generator", true)->value = PerimeterGeneratorType::Classic;` Comment this line out and you won't get the segfault.
> 
> I believe the code is attempting to enforce Classic (instead of Arachne) mode if you chose Spiral_Vase. The default, set in the top part of the code, is Classic, so I believe that unless you have specifically selected Arachne prior to this and then select Spiral_Vase you'd get some bad gcode, but that's just a guess on my part.
> 
> I'm not a c++ person by any means, so I don't know the appropriate way to set the value here, but if there is a good c++ programmer out there who understands this code they could probably fix it easily. As always, YMMV.

Tried without Arachne, that was indeed selected for me prior to vase. Crashes with classic selected too.
Also Using Super slicer 2.5.59.2 on Windows 10 (64bit).

EvilGremlin, 2023-08-09:
do not exist in https://github.com/supermerill/SuperSlicer/actions/runs/5548343816

fernplant, 2023-09-09:
> do not exist in https://github.com/supermerill/SuperSlicer/actions/runs/5548343816

Thanks very much. Installed nightly and can report that yes, it does fix the spiral vase mode bug

closed 2023-11-22 by supermerill