supermerill/SuperSlicer #3693: Arc Fitting?

foreachthing, 2023-03-06:
### What happened?

I must have missed something, because I'm not able to find arc fitting in Version 2.5.59.2.
In https://github.com/supermerill/SuperSlicer/issues/3683 I can see a screenshot ([here](https://user-images.githubusercontent.com/8691781/222904736-458936af-7f5d-4812-a2ec-ef1aeb97ef9f.png)) of this option but my downloaded version does not show it:

[image]

Thanks for your help.

### Project file & How to reproduce

Start SuperSlicer open Printer Settings -> General in Expert Mode.

### Version

2.5.59.2

### Operating system

Windows 11

### Printer model

any

vovodroid, 2023-03-06:
Are you sure you have correct version? Could you copy details from Help ->System Info?

foreachthing, 2023-03-07:
I am, in fact, very sure! :-)
Here's what I downloaded: https://github.com/supermerill/SuperSlicer/releases/download/2.5.59.2/SuperSlicer_2.5.59.2_win64_230304.zip

[image]

And still no Arc Fitting .... 
Any ideas are appreciated.

SuperSlicer
Version:   2.5.59.2
Build:     SuperSlicer_2.5.59.2

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22621), 64-bit edition
Total RAM size [MB]: 34,275MB
OpenGL installation
GL version:   4.6.0 NVIDIA 528.49
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
[... 330 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

neophyl, 2023-03-07:
I checked the same version I downloaded from following the link on the original release email, a few minutes ago, and its also not showing for me either.  
Windows 10 64 bit.

vovodroid, 2023-03-07:
Strange, I don't see any hiding this option in source code.

neophyl, 2023-03-07:
But there are many combinations of options for what to display and not in the gui.  

I just downloaded it at work and ran it there and also on Win10 and the options are showing on that machine.  

Could this be similar to the issue where the arachne options were not showing for some people ?

vovodroid, 2023-03-07:
> here are many combinations of options for what to display and not in the gui.

Yes, but it should be implemented in code, and I don't see such implementation.

foreachthing, 2023-03-07:
Huh?! I just downloaded the same version and started it on another computer (Win 10) and it's there, clear as day!
I'll try to export _this_ profile and use it on my other computer. So, in fact it is working! 👍🏼

foreachthing, 2023-03-07:
@neophyl sure!! If I could be of any help, let me know.

I'll try at Computer_1 again with a fresh profile (rename the current folder).
Computer_2 with a fresh profile folder works as expected.

neophyl, 2023-03-07:
I'll do the same after work.  On my work PC every single one of the 18 printer profiles installed is showing the addition of the G2/G3 arc fitting options.

My home pc on my normal printer profile doesn't.  I'll check the other installed ones at home later.

My work pc hasn't had as many different versions of SuSi installed as home, its mainly used to check out people issues on projects (Prusa forum and here).

foreachthing, 2023-03-07:
So, after removing all the profiles on computer_1, it works.
Now, for finding the conflicting setting(s) in any of the profiles.....

foreachthing, 2023-03-07:
Ok ... I may have found something!

I was using a custom ui_layout. Apperently some commands are new/old and ...  confuse each other.
After deactivating my custom ui_layout (set to standard) the option for G2/G3 becomes available.

So, what do you think, can I close this as I just need to adjust the `colors.ini` or is this a bigger deal?

neophyl, 2023-03-07:
Thanks for the hint.  All of my profiles on my home computer dont show the setting.  However I am not using a custom gui.  I have  "Standard: Default layout for superslicer" selected.  
However if I select the other option in the dropdown, "Example of another layout: More tags for ps/susi only settings, examples of new quick settings" then the settings do become visible on all my printers.

So the issue is the layout file but Ive only ever used defaults so I have no idea why the default one isnt showing them when selected.