supermerill/SuperSlicer #3697: [FR] Arc Fitting -> Travel moves

foreachthing, 2023-03-07:
**Describe the solution you'd like**
With the post-processing script from FormerLurker, there is an option to convert travel moves to G2/G3 as well: https://github.com/FormerLurker/ArcWelderLib#allow-travel-arcs

**Describe how it would work**
Have an option to enable travel moves as well or have them converted by default when Arc Fitting is enabled.
Models with lots of holes will profit from this setting. As travel moves without G2/G3 along holes are not easy on the printer - one can clearly hear the difference!
_(Same for skirt - brim is converted to arcs)_

**Describe alternatives you've considered**
Running FormerLurker's post-processing after exporting with Arc Fitting disabled.

Creo2005, 2023-03-07:
You don't need G2/G3 for a klipper, Kevin has said it 100 times. :)

foreachthing, 2023-03-07:
> You don't need G2/G3 for a klipper,

I have an Ultimaker 2 ... ☹️