supermerill/SuperSlicer #3702: Overhangs "thresholds for" option leads to incorrect Gcode in latest version

gifter77, 2023-03-08:
### What happened?

With option ticked (wrong):
[image]

Without option ticked (right):
[image]

Note that this option is on by default in at least Voron 2.4 profile.

### Project file & How to reproduce

[overhangs_threshold_bug.3mf.zip](https://github.com/supermerill/SuperSlicer/files/10920016/overhangs_threshold_bug.3mf.zip)

### Version

2.5.59.2

### Operating system

macOs

### Printer model

Voron 2.4

Cisien, 2023-03-09:
Nice find! I reported what turned out to be this here: https://github.com/supermerill/SuperSlicer/issues/3679

I wasn't sure what was causing it, but I can confirm that disabling this option removes my repro.

Grimones, 2023-03-15:
Thanks for the finding.

Tried setting `Bridge speed and fan` to 50% and it fixes the overhangs without turning off overhang detection on my model.