supermerill/SuperSlicer #3704: Supports going 2x speed under part

ssjbardock, 2023-03-08:
### What happened?

Underneath the floating part, supports are going 120mm/s, despite being set to 60mm/s (technically 50mm/s default speed with 120% set for support speed)

Can't tell why and can't find how to stop this.
[image]
[image]
[image]

### Project file & How to reproduce

Float a part or use raft
Supports under the part are going double speed of what is listed.

### Version

2.5.59.0

### Operating system

Windows 10

neophyl, 2023-03-08:
Please attach a zipped up project file that has this issue as requested when opening a new issue.  

For example if I use my normal profiles and add a box to a raft then I'm getting the correct speed for the raft as I have it configured.  This is why attaching a project file  is so important, its a snapshot of ALL your settings and makes it potentially possible to track down a cause.  Without one it is very difficult.

ssjbardock, 2023-03-08:
> Please attach a zipped up project file that has this issue as requested when opening a new issue.
>
> For example if I use my normal profiles and add a box to a raft then I'm getting the correct speed [...]

Updated original post with the project file 'too fast supports.zip'.

neophyl, 2023-03-08:
You are fast :)  uploaded while I was editing. see my last comment.

ssjbardock, 2023-03-08:
QUOTE: Edit - scratch that, I think its just treating the percentage value as a speed value. Change it to a different percentage and the speed appears to change to the same value in mm/s

I thought that may be the case, but then there's plenty of supports withing the object area all going 60mm/s.

neophyl, 2023-03-08:
Yeah but all of those are when the actual model is being printed.  The speed change only occurs ones it gets to the object.  
Btw you have some parts of the object with zero support that are getting printed on air.
[image]

neophyl, 2023-03-08:
[image]
Adding a box to the plater (not even part of the object) and the support speed starts to behave again.

ssjbardock, 2023-03-08:
That did not work for me unfortunately.

[image]

ssjbardock, 2023-03-08:
So it's not all empty layers, as I left some space between the cube top and part bottom, but it still kept the lower speed.

[image]

ssjbardock, 2023-03-08:
Well at least I can print with a workaround. Unless anyone else finds a magical 'empty layers sometimes ignore your setting' checkbox, pretty safely assume it's a bug.

Thanks for help testing!