supermerill/SuperSlicer #3705: Empty First Layer

old-relic, 2023-03-09:
### Version
2.5.59.0 Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
Windows 10 Pro version 22H2

### Behavior
* _Describe the problem_
* The slicer generates an empty layer 1
* _Steps needed to reproduce the problem_
* Just slice it
* _Expected Results_
* A good fuirst layer
* _Actual Results_
* An empty first layer
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
[Empty_layer_1.zip](https://github.com/su
[image]
permerill/SuperSlicer/files/10926992/Empty_layer_1.zip)
[image]

qwewer0, 2023-03-09:
I think that is the 0th layer in 0.00mm

old-relic, 2023-03-09:
Caused by selecting Marlin instead of Klipper in Printer settings. Solved

old-relic, 2023-03-09:
I am not sure that you are correct. If you look closely at the slider on
the right edge of the image of the build plate, you can see that it is
layer (1) at 0mm the model goes up to 3.80mm at layer (20)