supermerill/SuperSlicer #3710: Spiral mode is completely borked in 2.5.59.0

NovaViper, 2023-03-11:
### What happened?

I'm noticing SuperSlicer does not change the perimeters for vase mode, it keeps the perimeters the same instead of changing it to 1, which can and will completely break vase mode. and forces it to print like a normal object instead of vase mode

### Project file & How to reproduce

1. Load a model
2. Enable vase mode/spiral mode
3. Notice all other settings but perimeters are changed for vase mode

### Version

2.5.59.0

### Operating system

EndeavourOS with KDE 5.27.1

### Printer model

Creality Ender 3 with Klipper installed and BMG+V6 hotend+5015 cooling fan

paulgiordano, 2023-03-11:
So the problem was attempted to be fixed in 2.5.59.2 in src/libslic3r/PrintConfig.cpp with the addition of a line
`this->opt<ConfigOptionEnum<PerimeterGeneratorType>>("perimeter_generator", true)->value = PerimeterGeneratorType::Classic;`
inserted at around line 8057. This unfortunately causes a segfault. I believe that the purpose is to enforce Classic mode overriding Arachne mode if it's selected prior to this. At the beginning of PrintConfig.cpp there is code to set the default to Classic at around line 6110, 
`def->set_default_value(new ConfigOptionEnum<PerimeterGeneratorType>(PerimeterGeneratorType::Classic));`
This is all is exacerbated (I believe) by change #3204

NovaViper, 2023-03-11:
I can't really test this unfornately because of the package I'm using on ArchLinux. The package's an app image: https://aur.archlinux.org/packages/superslicer-prerelease-bin

paulgiordano, 2023-03-11:
https://drive.google.com/file/d/12BJpNsfmXRevX6LgGa9pNJd3VVYLHYJ0/view?usp=sharing
See if that'll work for you.

paulgiordano, 2023-03-11:
If you leave me an email address I will share a link to an appimage that I built for you with the "fix" (I commented the line out.)

CCS86, 2023-03-12:
For me 2.5.59.0 spiral vase works just fine, but the same project in 2.5.59.2 crashes (non-verbose)

paulgiordano, 2023-03-12:
Yeah, this fixes that.

NovaViper, 2023-03-12:
> https://drive.google.com/file/d/12BJpNsfmXRevX6LgGa9pNJd3VVYLHYJ0/view?usp=sharing See if that'll work for you.

The image works! But it still doesn't reduce the perimeters down to 1 like in v2.4.58.x does. For now, I'm gonna stick with the latest stable version til a patch comes out to address the segmentation faults in the dev builds 👍🏾

closed 2023-11-22 by supermerill