supermerill/SuperSlicer #3717: Presure Advance for filaments

ramaxa9, 2023-03-12:
Presure Advance for every filament like in BambooStudio please

NovaViper, 2023-03-12:
This is already possible actually. You can set the pressure advance value with gcode inside the filament specific gcode tab and it'll emit the gcode into the file so it sets the pressure advanced value at the beginning of the prints. I did this with Klipper by setting `SET_PRESSURE_ADVANCE ADVANCE=0.1492` inside the filament start gcode for my PETG and works perfectly for me 👍🏾

tomohulk, 2023-03-14:
Just as a tip, this is also very useful for setting you E rotation distance on a filament by filament bases as well:

```
;Filament gcode
SET_EXTRUDER_ROTATION_DISTANCE EXTRUDER=extruder DISTANCE=32.237091
```

closed 2023-03-14 by ramaxa9