supermerill/SuperSlicer #3720: Perimeters aren't generated properly on 2.5.59 release

HitLuca, 2023-03-13:
### What happened?

Looks like the latest pre-release has some issues dealing with perimeters in certain cases:

[image]

[image]

Tested the same stl using the latest officail release and the issue isn't visible there

### Project file & How to reproduce

[Hinge_Top_Right.zip](https://github.com/supermerill/SuperSlicer/files/10959591/Hinge_Top_Right.zip)

### Version

2.5.59

### Operating system

Windows 10

### Printer model

voron 0.2 with custom profile

JanB97, 2023-03-13:
[image]

Yes I also have issues with this. 
They just are in mid air for me without any reason for that.

Gizzzle, 2023-03-14:
[image]

Can confirm.

DiomedesDominguez, 2023-03-14:
Had the same issue last weekend, where a threaded part was creating a lot of inconsistency. Here is the preview and a test print
[image]

[image]

As you can see, the slicer was trying to print in mid air. I had to go back to 2.4 version.

mia-0, 2023-03-15:
Possible duplicate of #3681

DiomedesDominguez, 2023-03-19:
> Possible duplicate of #3681

it is

HitLuca, 2023-03-20:
Closing as duplicate of #3681 then

closed 2023-03-20 by HitLuca