supermerill/SuperSlicer #3726: Program crashes when switching to vase mode

pgsanjuan, 2023-03-14:
### What happened?

When selecting "Spiral vase" option in "Perimeters & Shell" section it shows the confirmation dialog with the configuration changes it is going to do. After accepting it, the program crashes without notice. It happens regardless the object as well as without object.

### Project file & How to reproduce

[mason-jar-vase-lid.zip](https://github.com/supermerill/SuperSlicer/files/10974711/mason-jar-vase-lid.zip)

### Version

Version 2.5.59

### Operating system

Microsoft Windows [Version 10.0.22621.1265]

### Printer model

Biqu B2 (B1 with dual extruder)

WestonWW, 2023-03-23:
I have the same issue.
Version 2.5.59.2
Win 10

shadowdro, 2023-04-05:
Same Issue. 

SuperSlicer
Version:   2.5.59.2
Build:     SuperSlicer_2.5.59.2

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 16,889MB
OpenGL installation
GL version:   3.0 Mesa 12.0.0-rc2
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits)
GLSL version: 1.30

dweenz, 2023-04-07:
Yup same here.

closed 2023-11-22 by supermerill