supermerill/SuperSlicer #3727: Seams on overhangs are non-continues when using threshold modifier for overhang speed and flow

Trini8ed, 2023-03-15:
### What happened?

When use the threshold modifier for overhangs, seams are incorrectly placed compared to when its disabled.

This issue is explicitly for how the seam becomes disjoint when encountering the modifier for overhangs. Please ignore the fact its not a straight line for aligned that is another issue.

### Project file & How to reproduce

1. Open attached STL in a new SS scene, make sure seams are aligned
2. Make sure these settings are enabled. under threshold for
[image]
4. Slice the part
5. Enable seams to show
6. Expect the following result
[image]

[Shape-Sphere.zip](https://github.com/supermerill/SuperSlicer/files/10975473/Shape-Sphere.zip)

### Version

2.4.58.4,2.5.59.2

### Operating system

Windows 11

### Printer model

prusa mk3s

pajtaz, 2024-04-12:
The aligned seam works in 2.5.59.2 on the one model I am testing but does not work on any version after that if the threshold modifier for overhangs is on. I really need a fix for this for the latest 2.5.59.8, it's driving me insane.

supermerill, 2024-05-03:
Can't reproduce, can I have your project file, and not only the stl?
[image]

Trini8ed, 2024-05-03:
Looks like this has been resolved as of 59.9, the main thing this ticket was about was making sure that when the overhang threshold was applied the seam locations would be the same as they were with it off.

[image]

pajtaz, 2024-05-12:
This is not resolved and in general there is a problem with overhang seams. I have attached 3MF file. You can see in the sliced preview that there is no seam shown on the external overhang wall while there is a seam shown on the external non-overhang wall. It is a huge problem when printing, as you can see in the third photo of the actual print with these settings. Seam is massive and ugly only on the overhang wall. I am using 30% overhang threshold so these walls would be printed slow enough and with enough cooling so they don't droop.
[image]
![Seam shown only on the other external non-overhang wall](https://github.com/supermerill/SuperSlicer/assets/19250632/9929158e-3d5d-
[Bad overhang seam.zip](https://github.com/supermerill/SuperSlicer/files/15285326/Bad.overhang.seam.zip)
45e5-843f-cab449d33474)
[image]

legend069, 2024-05-12:
seam placement should be avoided on curved corners like that as pressure advance tends to mess with it.

supermerill, 2024-05-15:
I changed the seam formula, so overhangs aren't evicted if the overhangs is very small or if there is only overhangs.
Seems like the behaviour is better. 
[image]
(huge seam gap to highlight them)

closed 2024-06-05 by supermerill

pajtaz, 2024-06-05:
Ok, I will test this at some point. For now I mitigated the problem by setting the overhang threshold do 30% instead of 25% (that made most previously identified overhangs as normal perimeters so this issue was not present).

Bunadaj801, 2024-06-15:
Hi, can I also emphasize how big an issue this is. I love ss and have switched from prusa but this issue is ruining my print quality. If I want control over my seams I have to disable overhang detection and have poor looking overhangs. So I have to choose, ugly seam placement or ugly overhangs. It's a shame as this isn't an issue in other slicers. Hope this gets resolved and would put this high on the priority list.