supermerill/SuperSlicer #3729: Enable visual indicators by default or allow an option to enable this by default

Trini8ed, 2023-03-15:
**Is your feature request related to a problem? Please describe.**
Slic3er seems to be fairly bad at seam-placement so its always good to be able to catch this at first glance without having to click that box each time as sometimes I forget to check this and think the seam placements are fine.

**Describe the solution you'd like**
I would like to have the show seams option in the GCODE preview to be enabled by default and not clicking the box each time to see where seam locations are. 
[image]

**Describe how it would work**
Either enable this feature by default doubt anyone would have any objections to this, if not or add an extra tic-box in preferences to enable this by default.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

seba9999, 2024-03-28:
Up ! 
I would love to see that too

vit-sh, 2024-07-31:
Up !
I would love to see that too