supermerill/SuperSlicer #3730: Prevent Temperature Commands From Being Injected Into Gcode Output

ctilley83, 2023-03-15:
**Is your feature request related to a problem? Please describe.**
The current way to prevent SuperSlicer from auto injecting temperature control is to set both the first layer and other layer to zero. This causes a problem for me because I'm sending those two parameters to a Duet Macro like the following. If I set those parameters to zero, then I can't send temps as parameters to my Duet macro.
```
M98 P"/macros/start.g" H[first_layer_temperature] B[first_layer_bed_temperature]
```
**Describe the solution you'd like**
The solution would be to add a checkbox to the temperature section that would prevent injecting temp control, but still be able to provide a values for temp fields. The tooltip in the screenshot below explains the current behavior.

[image]

**How to fix it**
should be pretty easy to modify the existing method and bind it to a checkbox. You can just remove the conditional that checks for 0 values in the temp input fields and bind the new checkbox to the existing method that prevents temp injection.

**Describe alternatives you've considered**
I've considered setting temps in my Filament configs on the Duet, but I modify temps often depending on the part I'm printing. That would lead to a lot of switching back and forth between Slicer and DWC. The current implementation is kind of an unnecessary roadblock for what I wanted to do.

legend069, 2023-03-16:
is this what you're looking for?
[image]

legend069, 2023-03-16:
> > is this what you're looking for?
> > 
> > [image]
> > 
> > 
> 
> Yes! What heathen put it there? Lol

Just make sure your print_start is setup correctly.
Copy paste the stuff super slicer auto injects if unsure.