supermerill/SuperSlicer #3731: Crash on Spiral Vase

sharonsolomon, 2023-03-17:
### What happened?

When I attempt to turn on Spiral Vase and accept the prompt to change settings to be compatible with that mode, the application crashes.  

I expected the settings to change and the application to stay open.

### Project file & How to reproduce

[vase failure.3mf.zip](https://github.com/supermerill/SuperSlicer/files/11002092/vase.failure.3mf.zip)

1. In print settings check 'Spiral Vase'
2. Click 'Yes' on the prompt to change settings required by spiral mode
Result: application crashes and closes 

### Version

2.5.59.2

### Operating system

macOS 13.2.1

### Printer model

Ender 3v2

workflow66, 2023-03-18:
Same here... No problems with the same STL in 2.5.59.0

Project file & How to reproduce

[VaseModeCrash.zip](https://github.com/supermerill/SuperSlicer/files/11009423/VaseModeCrash.zip)

    In print settings check 'Spiral Vase'
    Click 'Yes' on the prompt to change settings required by spiral mode
    Result: application crashes and closes

Version

2.5.59.2

Operating system

win10

Printer model

Ender 5 plus

oloendithas, 2023-03-18:
Same here.
Rolled back to 2.4.58.5_win64_220924

brownt12684, 2023-04-02:
Concur. Issue exists on any model where I enable vase mode.

matthewfallshaw, 2023-04-10:
Same here. Crashes even with an empty build plate.

Version: 2.5.59.2
OS: macOS 13.2.1

closed 2023-11-22 by supermerill