supermerill/SuperSlicer #3732: G02/G03 for Circular/Arc moves

daviedan, 2023-03-17:
**Is your feature request related to a problem? Please describe.**
I use superslicer to create GCODE for a hybrid manufacturing platform which runs on MACH4 Firmware. MACH4 (and MACH3) have the ability to read G02 (CW) and G03 (CCW) for arcs and circular moves. However, GCODE created using superslicer does not have this feature and instead discretises a circle into lots of small movements. This has been causing issues for my machine because of rapid acceleration and deceleration. 

**Describe the solution you'd like**
Ideally, I would like it if, when MACH3 is set as the firmware flavour, there was an option to use G02/G03 for arcs. 

**Describe how it would work**
I don't know how the slicer works to approximate an arc as discrete segments, but I'm sure it would be possible to recognise an arc and create the appropriate movements in G02/G03.

kursatu, 2023-03-18:
The latest nightly_dev branch does have the G2/G3 feature merged from BambuKab slicer.
It does have defects though. I've tried to use it and it generated incorrect toolpaths and extremely slow first layer
moves. I did not try much further. This is on Marlin 2, very recent build. 
You may want to check the discord channel and bring it up there if this is a very desired feature for you.

RiccaDS, 2025-03-06:
I'm interested too. With Mach3 smoothness and precision is not easy to accomplish if curves are approximated with segments.