supermerill/SuperSlicer #3741: Slicer generates one incorrect external perimeter

cfhsu, 2023-03-20:
### What happened?

On the secone layer, the slicer generates one straight external perimeter instead of a curve.
[image]
[image]

### Project file & How to reproduce

1. Download this mode from printable: [https://files.printables.com/media/prints/212570/stls/1948056_5abc6a2c-bc63-457c-865a-b9a9b77ace89/soap-dish-upper.3mf#_ga=2.79712919.1291359361.1678970532-102085863.1663576527](url)
2. Just slice it. Both Classic and Arachne generator produce this error.
3. [Soap Dish Upper - error.zip](https://github.com/supermerill/SuperSlicer/files/11013556/Soap.Dish.Upper.-.error.zip)

### Version

2.5.59.2

### Operating system

Windows 10

### Printer model

Voron 2.4r2

Creo2005, 2023-03-20:
Test your file. No problem with standard and Arachne to. Ss ver. 2.5.59.0
[image]

Creo2005, 2023-03-20:
> ### What happened?
> On the secone layer, the slicer generates one straight external perimeter instead of a curve. [image] [image]
> 
> ### Project file & How to reproduce
> 1. Download this mode from printable: [https://files.printables.com/media/prints/212570/stls/1948056_5abc6a2c-bc63-457c-865a-b9a9b77ace89/soap-dish-upper.3mf#_ga=2.79712919.1291359361.1678970532-102085863.1663576527](url)
> 2. Just slice it. Both Classic and Arachne generator produce this error.
> 3. Tried it on PrusaSlicer and it doesn't produce this error.
> 4. [Soap Dish Upper - error.zip](https://github.com/supermerill/SuperSlicer/files/11013556/Soap.Dish.Upper.-.error.zip)
> 
> ### Version
> 2.5.59.2
> 
> ### Operating system
> Windows 10
> 
> ### Printer model
> Voron 2.4r2

v. 2.5.59.2 is not yet ready to work

cfhsu, 2023-03-21:
Thank you very much Creo2005. Will use V2.5.59.0.

DVSVIDEO, 2023-04-05:
it is the same:
https://github.com/supermerill/SuperSlicer/issues/3771

mrulke, 2023-04-24:
this also happens on V2.5.59.0