supermerill/SuperSlicer #3749: Shut down after selecting Vase Mode

robblot, 2023-03-23:
### What happened?

The application shuts down after selecting Vase mode. this happen whether I have a file loaded or not.  

### Project file & How to reproduce

[cube-export.zip](https://github.com/supermerill/SuperSlicer/files/11055238/cube-export.zip)
After opening the file select vase mode and select yes. After a few seconds the program will shut down.

### Version

2.5.59.2

### Operating system

Windows 11

### Printer model

Sidewinder X1

manjaroblack, 2023-03-26:
I am having this issue as well.

dogmaphobic, 2023-03-27:
For what's worth:

```
Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   SuperSlicer                   	       0x104d1c6e9 Slic3r::DynamicPrintConfig::normalize_fdm() + 1753
1   SuperSlicer                   	       0x104cb709a Slic3r::Print::apply(Slic3r::Model const&, Slic3r::DynamicPrintConfig) + 442
2   SuperSlicer                   	       0x10519bb53 Slic3r::BackgroundSlicingProcess::apply(Slic3r::Model const&, Slic3r::DynamicPrintConfig const&, Slic3r::DynamicPrintConfig const*) + 163
3   SuperSlicer                   	       0x1053e7427 Slic3r::GUI::Plater::priv::update_background_process(bool, bool) + 215
4   SuperSlicer                   	       0x1054192f5 wxEventFunctorFunctor<wxEventTypeTag<wxTimerEvent>, Slic3r::GUI::Plater::priv::priv(Slic3r::GUI::Plater*, Slic3r::GUI::MainFrame*)::$_19>::operator()(wxEvtHandler*, wxEvent&) + 37
5   SuperSlicer                   	       0x10759bf3e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
6   SuperSlicer                   	       0x10759bcab wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
7   SuperSlicer                   	       0x10759bb69 wxEvtHandler::ProcessEvent(wxEvent&) + 41
8   SuperSlicer                   	       0x10759c0dc wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
9   SuperSlicer                   	       0x10757caaa wxTimerImpl::SendEvent() + 90
10  CoreFoundation                	    0x7ff806098b2d __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
11  CoreFoundation                	    0x7ff806098697 __CFRunLoopDoTimer + 820
12  CoreFoundation                	    0x7ff8060982b1 __CFRunLoopDoTimers + 285
13  CoreFoundation                	    0x7ff80607ed57 __CFRunLoopRun + 2142
14  CoreFoundation                	    0x7ff80607de7f CFRunLoopRunSpecific + 560
15  HIToolbox                     	    0x7ff80ff10766 RunCurrentEventLoopInMode + 292
16  HIToolbox                     	    0x7ff80ff10576 ReceiveNextEventCommon + 679
17  HIToolbox                     	    0x7ff80ff102b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
18  AppKit                        	    0x7ff8091092f3 _DPSNextEvent + 909
19  AppKit                        	    0x7ff809108174 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1219
20  AppKit                        	    0x7ff8090fa7b7 -[NSApplication run] + 586
21  SuperSlicer                   	       0x1074bc40e wxGUIEventLoop::OSXDoRun() + 174
22  SuperSlicer                   	       0x1075a5d8d wxCFEventLoop::DoRun() + 29
23  SuperSlicer                   	       0x10751b575 wxEventLoopBase::Run() + 85
24  SuperSlicer                   	       0x1074f3da4 wxAppConsoleBase::MainLoop() + 116
25  SuperSlicer                   	       0x1074815fa wxApp::OnRun() + 26
26  SuperSlicer                   	       0x10754e4f3 wxEntry(int&, wchar_t**) + 131
27  SuperSlicer                   	       0x10536fd1f Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 239
28  SuperSlicer                   	       0x10479f6ae Slic3r::CLI::run(int, char**) + 19358
29  SuperSlicer                   	       0x1047a3bc7 main + 375
30  dyld                          	       0x209c63310 start + 2432

```

Creo2005, 2023-03-27:
Version 2.5.59.2 has a lot of errors due to which the program does not work. I'm currently using 2.5.59.0

Carib3D, 2023-03-31:
I am also getting shutdowns on vase mode since this update.

toprobotmaster, 2023-03-31:
I am having this issue as well.

CCS86, 2023-04-01:
This has been reported over and over. Please search before reporting issues. The backlog is already enormous with merill out of action. If each issues gets duplicated 10 times, it's an even bigger nightmare to get through.

[image]

closed 2023-11-22 by supermerill