supermerill/SuperSlicer #3753: Error: Failed to load image from file "/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer".

id192313, 2023-03-25:
### What happened?

When opening SuperSlicer it gave and error. Can't close the error cause it kept popping up.
[image]

### Project file & How to reproduce

Install the latest version and overriding the previous version.

### Version

2.5.59

### Operating system

Mac OS Big Sur

### Printer model

Voron 2.4r2 Stealthburner

id192313, 2023-03-30:
Hi, still the same error.

Manfredo1952, 2023-06-11:
Installed 2.4.59.2 on MacOS 11.7.7 (Intel) today - same error message as reported aboce. Also after granting full access as proposed.  Any idea ?

skyaee, 2023-06-13:
Also after granting full access as proposed.err

devsfan1830, 2024-03-14:
Did you by chance copy a config from a Windows machine? I had this happen as well and was checking here before raising an issue. I mainly use a Windows laptop to slice, but I have a Mac in my basement to serve as my hobby bench machine and a BlueBubbles server (forwards messages between iMessage and my Pixel 7). Needed to slice something and I copied my config folder backup into my Mac version. I was met with that same unending error. I resolved it by using a text editor on the printer config to remove the paths for the build plate 3d model and overlay image.

andrewt27, 2024-09-09:
I managed to solve the problem. I restored the previous version of SuperSlicer and in the printer settings, I removed all the backgrounds that were highlighted in red (some corrupted PNG file).

[image]
[image]

Then I replaced SuperSlicer with the newer version, and it started without any errors.

SinisterRj, 2025-02-13:
Same problem here. It was working and them erro message that I can't get rid of.

SinisterRj, 2025-10-01:
My solution: if you are not able to open Susie to change the bed texture, look for your printer configuration file on "/Users/username/Library/Application Support/SuperSlicer/printer" and edit the file "your_printer.ini" and comment (adding #) to the following lines:
#bed_custom_model
#bed_custom_texture