supermerill/SuperSlicer #3759: Device tab to monitor print.

Stuckr, 2023-03-28:
I just tried out the Orca Slicer and it has a really useful feature the device tab which I can monitor my print from the slicer.  It appears to y a web browser extension that makes it possible to view my Mainsail interface directly in the slicer.   I imagine it would work with other web interfaces such as Fluidd, Octoprint, etc.     

Eliminates the need to go between browsers (which have many other tabs open most of the time.) and slicer. 

Being able to go to one place to slice and monitor prints would be very convenient.

This is the one feature that stood out and seemed very useful in Orcaslicer.

Thank you very much for all the hard work and the great product.
Robert

Glenno-H, 2023-03-29:
I found the same thing, its so convenient! (Other than Orca just having a better looking GUI IMHO)

sboehle, 2024-01-19:
@supermerill
This is a very helpfull feature and really convenient. After using Orcaslicer for a while, i miss this feature in SuperSlicer. I'm not a developer, but it should be more or less simple to implement. Perhaps using webview?

supermerill, 2024-01-23:
yes, possible.
I looked at the implementation from orca, and it's easy enough, but I need the 2.7 version stable running to be able to have the latest dependencies that contains the webview.