supermerill/SuperSlicer #3761: Steal PrusaSlicer 2.6 Alpha Organic Supports

rubyredrick, 2023-03-29:
**Is your feature request related to a problem? Please describe.**
Now that PrusaSlicer has implemented awesome tree supports in the current 2.6.0 alpha releases, I have to jump back and forth between SuperSlicer and PrusaSlicer. I really have come to love SS.
**Describe the solution you'd like**
Please incorporate PrusaSlicers Organic supports asap. Pretty Please.

ademlabs, 2023-03-29:
That would be really awesome! I second this.

LAP87, 2023-03-29:
Third!

trebory6, 2023-08-06:
> In the past Merill generally doesn't merge in changes from alpha or beta releases of PS. You are also merging in bugs that way. So Bumblebiber is quite correct. It makes zero sense doing so.

Well good thing it's no longer in Beta now! So let's revisit this question now the beta explanation is no longer valid!

j-snijder, 2023-08-07:
Do no merge PS 2.6 because of serious bug they will not help on.

Due to an update in the last 2 months PS causes the printbed of the Zonestar Z9V5 to go way down over its limits and crashes on the stepper motors which will damage your printer.

I have reported this several times but the devs on PS do not accept this bug report because, as they say, the update mentioned can not cause this because it only should have material updates.

I've checked it over and over a dozen times but IT IS IN THE UPDATE thats causing it.

So when you implement PS 2.6 you make your slicer unusable for all Z9V5 owners and thereby loose lots of good users.

Just a warnig, don't do it please, not before the bugs in PS is out of the slicer.

closed 2023-11-22 by supermerill