supermerill/SuperSlicer #3764: I would need to modify a code

onboarding92, 2023-03-30:
I would like to modify a code for more performance about my project.

THIS IS CODE NOW:
               **if (layer_time < slowdown_below_layer_time && fan_below_layer_time > 0) {
                    // Layer time very short. Enable the fan to a full throttle.
                    fan_speed_new = std::max(max_fan_speed, fan_speed_new);
                    bridge_fan_speed = std::max(max_fan_speed, bridge_fan_speed);
                    bridge_internal_fan_speed = std::max(max_fan_speed, bridge_internal_fan_speed);
                    ext_peri_fan_speed = std::max(max_fan_speed, ext_peri_fan_speed); // cannot be ovveridden**
                    
MY CODE:
                 
               **if (layer_time < slowdown_below_layer_time && fan_below_layer_time > 0) {
                    // Layer time very short. Enable the fan to a full throttle.
                    fan_speed_new = std::max(max_fan_speed, fan_speed_new);
                    bridge_fan_speed = fan_speed_new;
                    bridge_internal_fan_speed = fan_speed_new;
                    ext_peri_fan_speed = std::max(max_fan_speed, ext_peri_fan_speed); // cannot be ovveridden**
                    

how should i do? i create myself a build folder and do my build locally?

legend069, 2023-04-05:
what are you trying to accomplish ?

onboarding92, 2023-04-07:
With:

"bridge_fan_speed" and "internal_bridge_fan_speed" >(-1)
"autocooling" ON

IF

"bridge_fan_speed" and "internal_bridge_fan_speed" > "fan_speed" (resulting from autocooling interpolation) = "bridge_fan_speed" and "internal_bridge_fan_speed" are applied

IF

"bridge_fan_speed" and "internal_bridge_fan_speed" < "fan_speed" (resulting from autocooling interpolation) = "bridge_fan_speed" and "internal_bridge_fan_speed" are overridden

This behaviour is also described in the respective "bridge_fan_speed" and "internal_bridge_fan_speed" tooltips stating: "it won't slow down the fan if it's currently running at higher fan speed".

This default behaviour might not be an issue on small/low pressure fan printers, but using Hi Pressure fan styles like CPAP air pressure is so hi that bridges get pushed down making bridging effectively impossible [see picture]

onboarding92, 2023-04-07:
[image]

onboarding92, 2023-04-07:
What happens is if i print small parts fan_speed resulting from autocooling interpolation is very high and detrimental to bridges and first layer on top of support.

I modified cooling logic in such a way that "bridge_fan_speed" and "internal_bridge_fan_speed" are always applied.
Meaning that  "bridge_fan_speed" and "internal_bridge_fan_speed" are never overridden by fan_speed (resulting from autocooling interpolation).

A better way would be to put the option to activate and deactivate  "bridge_fan_speed" and "internal_bridge_fan_speed" overriding behaviour.
This way a better control over bridging and first layer on top of support can be achieved resulting in possible and longer bridges.

vovodroid, 2023-04-07:
Hi, so what is your question? How to build the project?