supermerill/SuperSlicer #3768: No Sparse Layers - Z doesnt move before changing filaments

utzy79, 2023-04-02:
[image]

When using no sparse layers, The z doesnt move to zero before changing the filament. 
Superslicer 2.5.59

mlee12382, 2023-11-27:
Here's more examples of the behavior. 

[image]

[image]

[image]

supermerill, 2023-11-27:
also buggy in prusaslicer 2.5
and seems disabled in prusaslicer 2.7