supermerill/SuperSlicer #3770: Monotonic infill is generated as rectilinear when infill_connection = Connected

justr0st, 2023-04-03:
### What happened?

Monotonic infill is generated in the same way as rectilinear when it is set to connected. It means, that infill changes it direction when moving around the holes. The issue is present for all kinds of infill:  solid, top, bottom and sparse (internal). It does not depend on which monotonic infill is selected (filled or not filled).

[image]

[image]

However, if infill_connection is changed to "Not Connected", it works as intended:
[image]

### Project file & How to reproduce

Open the 3MF project file and slice with top, bottom or solid monotonic infill set to "Connected".

[Test Part for Slicer.zip](https://github.com/supermerill/SuperSlicer/files/11136551/Test.Part.for.Slicer.zip)

### Version

2.5.59.2 Nightly https://github.com/supermerill/SuperSlicer/actions/runs/4532263472

### Operating system

Windows 11

NameOfTheDragon, 2023-12-27:
Bump @supermerill still present in 2.5.59.3

supermerill, 2023-12-28:
I wonder if I have to fix or not.
The "connect" will attach extrusion where they almost touch each other.
In this case, it attach two section together, and obviously, it's attached at the top, and so when extuding the whole thign it goes bot->top->bot.
The "not connected" is the thing to use for real monotonic.

Now, I can:
 . disable the "connection" setting when monotonic is selected and force not connected.
 . switch by default to "not connected" when selecting a monotonic (not a fan of this option)
 . let it as-is but set connection to disconnected by default for top, bottom and solid.
What do you think?

supermerill, 2023-12-28:
nvm, my previous message was just garbage.
I'll fix.

supermerill, 2023-12-28:
suprisingly time-consuming to find the problem. 4 hours. Still, a one-line fix in 5 minutes when the problem was identified.